ISPC-TST-EM2022/Grupo2 #18: PRACTICA IOT : Ejercicio #1 : PUNTO 5 Y 6 -

WayarCarla, 2022-08-21:


NataliaGalliani, 2022-08-21:
Ejercicio #1
Utilizar el terminal virtual del simulador para realizar los siguientes ejercicios:
**Clase: Serial**
Métodos: read(), write(), parseInt(), parseFloat(), available(), print(), println() y begin().

1.    Pedir el ingreso de dos números por terminal y mostrar la suma por la pantalla.  
2.    Preguntar Nombre, Apellido, Direccion y Celular en mensajes diferentes. Completar las respuestas.
3.    Realizar la suma de n números reales ingresados por terminal y calcular su media.
4.    Pedir un valor entre 0 y 255, para controlar el brillo de un diodo led.

**5.    Ingresar una secuencia de 8 bits por terminal , que activaran los leds correspondientes por el puerto D.** 

**6.    Realizar una función bool cerradura({tipo} clave); que retorne true cuando la clave numérica sea 2147483648. Elija el “tipo” de dato apropiadamente. Implemente el circuito que active un rele; Utilice el terminal virtual.**

WayarCarla, 2022-08-28:
FIN

closed 2022-08-28 by WayarCarla